OpenVnmrJ/OpenVnmrJ #677: How are FIDs stored

santimirandarp, 2022-02-20:
There are a few sentences that _seem_ to be contradictory in the VnmrJ User Programming Manual. It'd be great if you give me some hints. This is what I found:

> Binary format
  (...) data stored in the file followed by the spectral data in integer or floating point format.  

And 
> FIDs are stored as either 16- or 32-bit integer binary data files, depending on whether the 
data acquisition was performed with dp='n' or  dp='y', respectively.

And
> The actual FID data is typically stored as pairs floating-point numbers.

From my tests, I looks like FIDs are normally Float32, but not always. 

How should we interpret this?

DanIverson, 2022-02-23:
It depends on the instrument used to collect the data. On DDR and DD2 systems, FID data is float. All other systems collect data generally as 32-bit integer although 16-bit integer is also supported. OVJ can handle all three types. The status field of the file header specifies the type of data with the S_32 and S_FLOAT bits.

santimirandarp, 2022-02-23:
That's of great help thank you for taking the time in replying.

And the last bit: those are normally complex numbers interleaved real,imaginary,real1, img1... Right?

I was expecting the 'is complex' file header to indicate this, but most of them are is complex=0 meaning real and the fid that array of complex numbers described above.

Sent from ProtonMail mobile

-------- Original Message --------

DanIverson, 2022-02-23:
Yes, RE1, IM1, RE2, IM2 ...
If you are looking at the output of the ddff command, the status field is being displayed as a Hex number. I should probably display a 0x in front of it. When I rt('fid1d') from fidlib, the status is displayed as 99, meaning the following bits are on: S_DATA, S_FLOAT, S_COMPLEX, and S_DDR

santimirandarp, 2022-02-23:
I am writing a small JS parser for fids, but I don't have much experience with the OvnmrJ, though I installed it in Ubuntu to help guide my work, as a reference, and also read part of the C code that you wrote.

It reads 1Ds and traced 1Ds (I interpret those as arrays of 1Ds), but am not yet sure how to tell whether data points are real or complex numbers (ATM I am guessing all 1Dfids are complex), but as said isComplex param is normally 0.

If you have anytime at some point I could make a few other Qs (particularly for 2Ds), but thanks for the replies otherwise.

-------- Original Message --------

DanIverson, 2022-02-24:
Here is a python script to read FIDs. ( https://github.com/DNPLab/DNPLab/blob/master/dnplab/io/vnmrj.py ). Maybe it will help.

santimirandarp, 2022-02-24:
Looks promising. If there is something I could help here let me know. I am interested.

Thanks for the tips.

closed 2022-02-24 by santimirandarp