OpenVnmrJ/OpenVnmrJ #729: Installation on WSL2 (Windows 11)

darekb94, 2022-11-15:
> 

> It's working now! WSL2 was required. It will be officially available in April 2020, but everyone can decide to join the Insider preview which offers updates in advance. Thank you!!

Recently, I have tried to install OpenVnmrJ 3.1 on WSL2 on Windows 11 but unfortunately I encountered the problems similar to those described earlier in another issues. I would like to use this program for data analysis only so I did not configure it as a spectrometer. The problems are as follows:
-  after launching OpenVnmrJ, a window opened for a while, but then I returned error "Vnmrbg could not be started" and the program had to be closed down
- I could not invoke Vnmrbg from the terminal (allthoug it exists in the folder vnmr/bin) due to an error "No such file or directory" whereas a command ldd Vnmrbg returns "       not a dynamic executable" (with many spaces at the beginning)
- ovjDiag returns the same error: "       not a dynamic executable" (with many spaces at the beginning)
- the file VnmrjMsgLog ends with the following error:

"makeDBConnection failed on <here my hostname> port 5432,
be sure database exists and that vnmr1 has access to it.
Try having the system administrator execute the command 'createuser vnmr1'.
Also check that if a special port number  is used for postgres,
that this user has the environment variable 'PGPORT' set  correctly.
Is it possible that a specific error can be found in the log file:
/vnmr/pgsql/pgsql.log "
- I tried to invoke 'createuser vnmr1' but I obtained the following error:

"createuser: error: connection to server on socket "/var/run/postgresql/.s.PGSQL.5432" failed: No such file or directory
        Is the server running locally and accepting connections on that socket?"
- the file '/vnmr/pgsql/pgsql.log' has the following content:

"LOG:  database system was shut down at 2022-11-12 20:51:46 CET
LOG:  MultiXact member wraparound protections are now enabled
LOG:  database system is ready to accept connections
LOG:  autovacuum launcher started
LOG:  autovacuum launcher shutting down
LOG:  received smart shutdown request
LOG:  received immediate shutdown request
WARNING:  terminating connection because of crash of another server process
DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
HINT:  In a moment you should be able to reconnect to the database and repeat your command.
LOG:  database system is shut down
LOG:  database system was shut down at 2022-11-13 15:10:54 CET
LOG:  MultiXact member wraparound protections are now enabled
LOG:  database system is ready to accept connections
LOG:  autovacuum launcher started
ERROR:  role "walkup" already exists
STATEMENT:  CREATE ROLE walkup NOSUPERUSER CREATEDB NOCREATEROLE INHERIT LOGIN;
ERROR:  role "service" already exists
STATEMENT:  CREATE ROLE service NOSUPERUSER CREATEDB NOCREATEROLE INHERIT LOGIN;"

Can anyone find a solution to these problems?
Is it really possible to run OpenVnmrJ on WSL2?

vm-chem, 2023-01-23:
I am in the exact same use-case scenario.

> after launching OpenVnmrJ, a window opened for a while, but then I returned error "Vnmrbg could not be started" and the program had to be closed down
I could not invoke Vnmrbg from the terminal (allthoug it exists in the folder vnmr/bin) due to an error "No such file or directory" whereas a command ldd Vnmrbg returns " not a dynamic executable" (with many spaces at the beginning)
ovjDiag returns the same error: " not a dynamic executable" (with many spaces at the beginning)

Assuming you are using Ubuntu, I solved this by installing gcc-multilib for the cross-compiling, as OpenVnmrJ is a 32-bit executable:

`sudo apt install gcc-multilib`

>"makeDBConnection failed on port 5432,
be sure database exists and that vnmr1 has access to it.
Try having the system administrator execute the command 'createuser vnmr1'.
Also check that if a special port number is used for postgres,
that this user has the environment variable 'PGPORT' set correctly.
Is it possible that a specific error can be found in the log file:
/vnmr/pgsql/pgsql.log "

I have the same error, but I didn't solve this one yet.

DanIverson, 2023-01-23:
As the OVJ admin (typically user vnmr1), run the command
  dbsetup
Alternatively, many people just turn the pgsql database off. You can do this by running
  vnmrj admin
and then selecting Configure->Locator Config...  and then checking the "Locator Off" checkbox.