OpenVnmrJ/OpenVnmrJ #734: Error ownloading files to the acquisition console

rmserra2, 2022-12-15:
Describe the bug
During the installation of  OVJ 3.1 on my spectrometer console. I get an error during the downloading 14 files to the acquisition console

To Reproduce
    As vnmr1, unzip the DDR file downloaded from github
    run ./load.nmr
    After the installation of openvnmrj and network configuration, during the download of the forth file to the acquisition console an error says download failed.
 I've tried to reboot the console and run setacq several times but I get the same error

Expected behavior
The complete download of the 14 files and starting the spectrometer mode .

Version of OpenVnmrJ you are using.
OpenVnmrJ 3.1 downloaded from GitHub on Dec, 1st, 2022.

Messages and Logs
Testing console software version, will try up to 20 seconds
downloading 14 files to the acquisition console. 
download file 1 of 14
download file 2 of 14
download file 3 of 14
download file 4 of 14
download failed.

Environment (please complete the following information):
    OS: centos 7.9
    hostnamectl gives 
    CPE OS Name: cpe:/o:centos:centos:7
     Kernel: Linux 3.10.0-1160.el7.x86_64
     Architecture: x86-64
     Console Varian NMR system

DanIverson, 2022-12-17:
There have been a couple of fixes to the setacq script. Github won't let me attach the file. If you send your email address, I will forward it to you. Copy it to /vnmr/bin/setacq and try then run setacq. When it asks if you would like to re-configure the spectrometer, answer y.

DanIverson, 2022-12-17:
[setacq.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/10252313/setacq.txt)

Github allowed me to attach setacq if it has a .txt extension. After downloading it, as vnmr1, do
  cp setacq.txt /vnmr/bin/setacq
  chmod 755 /vnmr/bin/setacq

DanIverson, 2023-01-23:
Assuming the new setacq fixed the problem. If not, please re-open the issue.

closed 2023-01-23 by DanIverson