OpenVnmrJ/OpenVnmrJ #745: Ambiguous status of NSUFFIX constant

bdezonia, 2023-02-28:
The constant NSUFFIX is defined in both soliddefs.h and solidmisc.h. In soliddefs.h it is equal to MAXSTR (so probably 256). In solidmisc.h it is equal to 13. This could be a source of bugs. Especially depending on which include or in what order both includes show up for various classes. As an aside seqName in pboxpulse.h is hard coded at 13 and does not use an NSUFFIX definition. We should investigate what the best thing to do for all of this is. Note I am willing to help with this investigation if necessary.

bdezonia, 2023-02-28:
FYI here is a grep of all the places in the code NSUFFIX is set:

bluejay:~/dev/OpenVnmrJ> grep -r "define.*NSUFFIX" *
src/vjqa/vjtest/hh2dhdec/psg.orig/soliddefs.h:#define NSUFFIX 13
src/nvpsg/soliddefs.h:#define NSUFFIX MAXSTR
src/nvpsg/solidmisc.h:#define NSUFFIX 13

DanIverson, 2023-02-28:
Should be fixed with PR #751

closed 2023-02-28 by DanIverson