OpenVnmrJ/OpenVnmrJ #750: Ambiguous status of MAXSTR constant

bdezonia, 2023-02-28:
In Pbox_psg.h MAXSTR can be defined as 512:

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/master/src/psg/Pbox_psg.h#L17

This file is included in many places. But it is also included in solidstandard.h in /src/nvpsg.

Many of the files in /src/nvpsg use MAXSTR. And almost everywhere it is defined as 256. The base one seems to be here:

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/c306df2964c3a823588757772f9f5260722b46e1/src/nvpsg/acqparms.h#L51

There are numerous other files in /src/nvpsg that set it to 256 if it is not defined.

One issue with these definitions is that depending upon the existence of an include or a different ordering of the include files MAXSTR can be different values in different files (or in different parts of files). This can lead to numerous hard to debug problems.

I am willing to help investigate this if needed.

bdezonia, 2023-02-28:
Things are messier than I thought:

bluejay:~/dev/OpenVnmrJ> grep -r "define.*MAXSTR" *
src/bootpd.rh51/bootpd.h:#define MAXSTRINGLEN		 80	/* Max string length */
src/bin/Pbox_def.h:#define MAXSTR  256
src/bin/Pxspy.c:#define MAXSTR 512
src/bin/Pxsim.c:#define MAXSTR 512
src/bin/vnmrio.h:#define MAXSTR 1024
src/bin/read_raw_data.c:#define  MAXSTR  256
src/bin/Pxfid.c:#define MAXSTR 512
src/Asp/AspUtil.h:#define MAXSTR2  512 
src/bin_image/main_fdf2nifti.c:#define MAXSTR 1024
src/bin_image/main_fdf2avw.c:#define MAXSTR 1024
src/bin_image/utils.h:#define MAXSTR 1024
src/vnmr/acqhdl.c:#define MAXSTR 	256
src/vnmr/set3dproc.c:#define MAXSTR		256
src/vnmr/asmfuncs.c:#define MAXSTR  256
src/vnmr/asmfuncs.c:#define MAXSTR 256
src/vnmr/vnmrsys.h:#define MAXSTR  256
src/vnmr/writeTrash.c:#define MAXSTR 512
src/vnmr/acqfuncs.c:#define MAXSTR 256
src/vnmr/vnmrMD5.c:#define MAXSTR	512 
src/vnmr/weight.c:#define MAXSTR		256
src/vnmr/p11Tools.c:#define MAXSTR 512
src/vnmr/chchsums.c:#define MAXSTR	512 
src/vnmr/writeAaudit.c:#define MAXSTR 512
src/vnmr/lp.c:#define MAXSTR		256
src/vnmr/specfreq.c:#define MAXSTR 256
src/vnmr/auditcp.c:#define MAXSTR  1024
src/vnmr/md5c.c:#define MAXSTR 512
src/vnmr/ft2d.c:#define MAXSTR		256
src/vnmr/chVJlist.c:#define MAXSTR 512
src/vnmr/lockfreqfunc.c:#define	MAXSTR	256
src/vnmr/ft.c:#define MAXSTR		256
src/kpsg/PboxM_psg.h:#define MAXSTR 512
src/kpsg/meat.c:#define	MAXSTR	28
src/kpsg/meat.c:#define	MAXSTR_UPLUS  256
src/kpsg/lc_hdl.c:#define MAXSTR 256
src/kpsg/Pbox_psg.h.old:#define MAXSTR 512
src/kpsg/acqparms2.h:#define MAXSTR 256
src/kpsg/array.c:#define MAXSTR		256
src/kpsg/psg.c:#define MAXSTR		256
src/psg/acqparms.h:#define MAXSTR 256
src/psg/cmed.h:#define MAXSTR  256           /* maximum characters per string */
src/psg/psg4dps.c.bk:#define MAXSTR 256
src/psg/ssha.c:#define MAXSTR 256
src/psg/cps.c:#define MAXSTR 256
src/psg/lc_hdl.c:#define MAXSTR 256
src/psg/Pbox_psg.h:#define MAXSTR 512
src/psg/psg4dps.c:#define MAXSTR 256
src/psg/abort.c:#define MAXSTR 256
src/psg/arrayfuncs.c:#define MAXSTR 256
src/psg/ecc_worstCase.c:#define MAXSTR 256
src/psg/convertipa.c:#define MAXSTRING	32
src/psg/psg_utils.h:#define MAXSTR  256           /* maximum characters per string */
src/psg/freq_device.c:#define	MAXSTR	256
src/psg/treefuncs.c:#define MAXSTR 256
src/psg/lintfile.c:#define MAXSTR 256
src/psg/psg.c:#define MAXSTR 256
src/ib/params.h:#define MAXSTR  1024    /* Maximum length of string values, including EOS. */
src/3D/fid3d.c:#define MAXSTR		256
src/3D/plextract.c:#define MAXSTR		256
src/ncomm/ipcKeyDbm.h:#define IPCKEY_MAXSTR_LEN 256
src/nvpsg/Shim.cpp:#define	MAXSTR	256
src/nvpsg/acqparms.h:#define MAXSTR 256
src/nvpsg/Console.cpp:#define MAXSTR 256
src/nvpsg/cps.c:#define MAXSTR 256
src/nvpsg/lc_hdl.c:#define MAXSTR 256
src/nvpsg/cpsg.h:#define MAXSTR 256
src/nvpsg/soliddefs.h:#define NSUFFIX MAXSTR
src/nvpsg/psgmain.cpp:#define MAXSTR 256
src/nvpsg/arrayfuncs.c:#define MAXSTR 256
src/nvpsg/GradientBase.cpp:#define MAXSTR 256
src/nvpsg/DDRController.cpp:#define MAXSTR     256  // needed for getstr
src/nvpsg/MasterController.cpp:#define MAXSTR	256
src/nvpsg/GradientBridge.cpp:#define MAXSTR 256
src/aip/aipRQnode.h:#define MAXSTR  256

bdezonia, 2023-03-03:
BTW if you might want to wait a while before addressing this. I made (a lot) of local changes on a branch I have that automatically checks destination buffer sizes so that they never get stuffed past their bounds throwing a warning or error when it happens. This code works for all calls (we desire) to strcpy(), strcat(), and sprintf(). It works via preprocessor macros. I was trying to use it to check for buffer overwrite problems. My code only applies to /src/nvpsg but you could use it to safely be able to define MAXSTR (or local MAXSTRs) in various ways and catch errors early. Maybe we talk about this when I am back from vacation.

DanIverson, 2023-03-03:
Newer gcc compilers give warnings when there are potential buffer overflows. (-Wformat-overflow). I have been (slowly) fixing these as I make changes in .c files for other reasons.