OpenVnmrJ/OpenVnmrJ #772: dicom files are empty  in the output dicom directory in openvnmrj 3.1

mosquitozheng, 2023-05-16:
Openvnmrj 3.1 cannot properly convert FDF images to Dicom. The output directory are all empty with nothing (no Dicom files created). This function was working well in the Agilent Vnmrj. The binary file for this conversion is FdfToDcm. Don't know how to solve this bug

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'I/O' -->'Format Conversion'-->choose "format: DICOM"--> chose a loaded image
2. Click on 'Save'
3. It would show: "save ......Done", but not files in the output folder created.

DanIverson, 2023-05-16:
The attached zip file contains an aipOut script that corrects the grep error. Unzip the attachment and then
  mv aipOut /vnmr/bin/.
Does this correct the DICOM conversion?
[fix.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/11490735/fix.zip)

mosquitozheng, 2023-05-17:
> The attached zip file contains an aipOut script that corrects the grep error. Unzip the attachment and then mv aipOut /vnmr/bin/. Does this correct the DICOM conversion? [fix.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/11490735/fix.zip)

Comment: I did remove the above "grep ....." error. But the output dicom folder is still empty.

mosquitozheng, 2023-05-17:
[image]

Comment：remove the above 'grep -o...' in two places of aipOut, it will generate the dicom files.
[image]

mosquitozheng, 2023-05-17:
If I remove the "grep ....' in the aipOut, openvnmrj can convert individual image folder to dicom. However, if I choose a whole study (which has several folders of img FDFs from different sequences, it does nothing. It works well under redhat 6.3, but not on  openvnmrj under centos 7. please refer to below pictures.
[image]
[image]

closed 2023-07-11 by DanIverson