OpenVnmrJ/OpenVnmrJ #785: One definition of M_PI is weird

bdezonia, 2023-07-16:
I am browsing source code to learn a bit about the scope of the project. Anyhow in a number of places there is code to define M_PI if it does not exist. That code basically looks like this:

#ifndef M_PI
#define M_PI    3.14159265358979323846
#endif

However on this line of code it is different:

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/9e880791cdd2222986e86426d5c8d665528417ca/src/vwacq/autolock.c#L3063

Is that because this is related to code we have no control over? Or was an edit missed?

DanIverson, 2023-07-17:
Answer is lost in the mists of time. Probably because vwacq is the code that runs on the Inova console, which is a 16-bit CPU. The vwacq code is not compiled as part of OVJ. The Inova console executable (vxBoot), based on the vwacq sources, is taken from the VnmrJ 4.2 release.

bdezonia, 2023-08-14:
Dan. I think we can just close this.

closed 2023-08-14 by bdezonia