OpenVnmrJ/OpenVnmrJ #813: Infostat -l throws odd values for vt and spin

JessiBaughman, 2023-09-02:
**Describe the bug**
When running ```Infostat -l``` or ```Infostat -log /path/to/dir``` it will populate event.xml and lock.xml correctly, but throws frequent incorrect values for vtval and vtset in vt.xml and spinset and spinval in spin.xml. Running ```Infostat```, ```Infostat -d```, or ```showstat``` never shows incorrect values.
Additionally, ```Infostat -l``` doesn't automatically start with ```Infostat -port ####``` on Mercury/Inova consoles (does start on NMRS/DDR2).

**Version of OpenVnmrJ you are using.**
All built on RHEL 8
 - Mercury
   - MI #732
 - Inova
   - MI  #714
 - NMRS
   - #728
 - DDR2
   - #718

**Messages and Logs**
From the NMRS system regulated at 30 C and 20 Hz spinning.
```
vnmr1@nmrs500-01 logs]$ tail vt.xml 
<vtval value="29.9 C" time="1693622325"/>
<vtset value=" 0.0 C" time="1693622336"/>
<vtval value="3274.5 C" time="1693622336"/>
<vtset value="1233.8 C" time="1693622337"/>
<vtval value="1310.6 C" time="1693622337"/>
<vtset value="30.0 C" time="1693622337"/>
<vtval value="3274.5 C" time="1693622337"/>
<vtval value="29.9 C" time="1693622337"/>
<vtval value="30.0 C" time="1693622337"/>
</statlog>
```
```
[vnmr1@nmrs500-01 logs]$ tail spin.xml 
<spinset value="288 Hz" time="1693622366"/>
<spinset value="20 Hz" time="1693622366"/>
<spinval value="0 Hz" time="1693622367"/>
<spinset value="288 Hz" time="1693622373"/>
<spinset value="20 Hz" time="1693622373"/>
<spinval value="0 Hz" time="1693622373"/>
<spinset value="288 Hz" time="1693622376"/>
<spinset value="20 Hz" time="1693622377"/>
<spinval value="0 Hz" time="1693622377"/>
</statlog>

```

**Environment (please complete the following information):**
All systems tested are the same:
```
           .MMM..:MMMMMMM                  vnmr1@nmrs500-01 
          MMMMMMMMMMMMMMMMMM               ---------- 
          MMMMMMMMMMMMMMMMMMMM.            OS: Red Hat Enterprise Linux 8.8 (Ootpa) x86_64 
         MMMMMMMMMMMMMMMMMMMMMM            Host: HP Z420 Workstation 
        ,MMMMMMMMMMMMMMMMMMMMMM:           Kernel: 4.18.0-477.21.1.el8_8.x86_64 
        MMMMMMMMMMMMMMMMMMMMMMMM           Uptime: 16 days, 10 hours, 19 mins 
  .MMMM'  MMMMMMMMMMMMMMMMMMMMMM           Packages: 1926 (rpm), 27 (flatpak) 
 MMMMMM    `MMMMMMMMMMMMMMMMMMMM.          Shell: bash 4.4.20 
MMMMMMMM      MMMMMMMMMMMMMMMMMM .         Resolution: 1920x1200 
MMMMMMMMM.       `MMMMMMMMMMMMM' MM.       DE: GNOME 3.32.2 
MMMMMMMMMMM.                     MMMM      WM: Mutter 
`MMMMMMMMMMMMM.                 ,MMMMM.    WM Theme: Adwaita 
 `MMMMMMMMMMMMMMMMM.          ,MMMMMMMM.   Theme: Adwaita [GTK2/3] 
    MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM   Icons: Adwaita [GTK2/3] 
      MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM:   Terminal: gnome-terminal 
         MMMMMMMMMMMMMMMMMMMMMMMMMMMMMM    CPU: Intel Xeon E5-1620 0 (8) @ 3.800GHz 
            `MMMMMMMMMMMMMMMMMMMMMMMM:     GPU: NVIDIA Quadro 2000 
                ``MMMMMMMMMMMMMMMMM'       Memory: 2531MiB / 7611MiB 
```

JessiBaughman, 2023-09-02:
Perhaps related to #296?

JessiBaughman, 2023-09-02:
I'm not sure if it's related, but the user log (via the Status Charts) works unless you load the system log. Then the set point gets thrown off, but the actual value continues to register correctly. But the user log stops on operator logout, so I'd prefer using ```Infostat -l```.

DanIverson, 2023-10-06:
Hi Jessi,
I have been trying to reproduce this but have been unable to do so. It is curious that multiple values are recorded with the same time= value. Could you start the Infostat with
  strace -o /vnmr/tmp/infobug -s 128 Infostat -l
After some of the bad values are recorded, kill the process and send me the /vnmr/tmp/infobug file.

JessiBaughman, 2023-10-07:
[infobug.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/12838126/infobug.txt)

The attached file was created with an Inova system running PR build #714. VT was set to -18.8 with 20 Hz spinning and a lock level around 87.

JessiBaughman, 2023-10-07:
Looks like a lot of missing libraries mentioned at the top.  All the mentioned libraries (libm.so.6, libtirpc.so.3, libc.so.6, libgssapi_krb5.so.2, libkrb5.so.3, libk5crypto.so.3, libcom_err.so.2, libpthread.so.0, libdl.so.2, libkeyutils.so.1, libcrypto.so.1.1, libresolv.so.2, libselinux.so.1, libz.so.1, libpcre2-8.so.0) are present in both /lib and /lib64.

DanIverson, 2023-10-24:
Should be fixed with PR #824

closed 2023-10-24 by DanIverson