OpenVnmrJ/OpenVnmrJ #846: Possible bugs in _pboxsimpulse() in pboxpulse.h

bdezonia, 2024-02-09:
I have been studying the _pboxsimpulse() code in pboxpulse.h. I have come across a few lines that look fishy to me. I think there are appearances of calls that are not appropriate. I will provide code links with explanations.

The call to decunblank() here does not make sense since we only every obsblank() and dec2blank() in the surrounding code subclause.

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/0ad7471b90726f53a32eef7d3aa53085f84d9bf1/src/nvpsg/pboxpulse.h#L626

The call to obsunblank() here does not make sense since we only ever decblank() and dec2blank() in the surrounding code subclause.

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/0ad7471b90726f53a32eef7d3aa53085f84d9bf1/src/nvpsg/pboxpulse.h#L678

The call to decunblank() here does not make sense since we only every obsblank() and dec2blank() in the surrounding code subclause.

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/0ad7471b90726f53a32eef7d3aa53085f84d9bf1/src/nvpsg/pboxpulse.h#L712

The call to obsunblank() here does not make sense since we only every decblank() and dec2blank() in the surrounding code subclause.

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/0ad7471b90726f53a32eef7d3aa53085f84d9bf1/src/nvpsg/pboxpulse.h#L744

bdezonia, 2024-02-10:
I've noticed that these unblank calls only happen when channel 3 is involved.

When (ch1 == 1 && ch2 == 3) || (ch1 == 3 && ch2 == 1) the code does the decunblank()

When (ch1 == 2 && ch2 == 3) || (ch1 == 3 && ch2 == 2) the code does the obsunblank()

I think this make very little sense but thought I should mention the pattern.

bdezonia, 2024-02-10:
Another set of troubling code is the order of operations that happen at the end of the three impulse routines.

The last few lines of any subcase in the one channel routine basically look like this:
         
         pulse();
         delay(some time);
         chan_unblank();

The last few lines of any subcase in the two channel routine basically look like this:

         pulse();
         chan1_unblank();
         chan2_unblank();
         delay(some time);

The last few lines of any subcase in the three channel routine basically look like this:

         pulse();
         chan1_unblank();
         chan2_unblank();
         chan3_unblank();
         delay(some time);

So my question is why does the one channel case do delay first and then unblank while the two and three channel routines do the unblank and then the delay last? Shouldn't they be defined consistently?

bdezonia, 2024-02-10:
Back to the four unexpected unblank() calls maybe sim3_shaped_pulse blanks all three channels and unblanking is in order? Or the sim3_shaped_pulse code was cut from the 3 channel routine when writing the 2 channel routine and people forgot to delete the unnecessary unblank()'s.

bdezonia, 2024-02-13:
For the cut and paste idea of he sim3 pulse code I am now realizing that the the two channel code calls sim3_pulse() instead of sim2_pulse(). Maybe this is the real bug (that would also eliminate the extra unblanking).

bdezonia, 2024-03-04:
One more possible bug found. On the line below there is a delay() call using unknown variable rof1 (which must somehow be visible to the file since it compiles fine). This is in _pboxsim3pulse() and is the only one of the six subcases of pulsing that has that delay() call at all. Any idea why?

https://github.com/OpenVnmrJ/OpenVnmrJ/blob/5ae038f9eaeace49d1744c06134f0d8bffc0129d/src/nvpsg/pboxpulse.h#L836

bdezonia, 2024-03-05:
Also, for completeness sake, there is an active pull request for one more flaw. See here:

https://github.com/OpenVnmrJ/OpenVnmrJ/pull/845

DanIverson, 2024-11-14:
Give me a little time to review your comments.

bdezonia, 2024-12-18:
In the case of whether the single channel pulse code is right [in terms of ending with delay() and then unblank()] or whether the two and three channel pulse code is right [ending with unblank() and then delay()]:

I talked to a scientist that admittedly doesn't know the code but their thought was that you always want to pulse / delay / unblank because the alternative (unblanking and then delaying to do the pulse) might hatch artifacts in the signal. Again, this scientist would not swear to it but it gives us something to go on.

bdezonia, 2024-12-30:
Hey Dan. I am looking a little at the code in pboxpulse.h to see if I can determine what should be a good approach to making sure all the impulse routines (numch == 1 or 2 or 3) behave similarly. Partially to figure out the pulsing logic. I have found that the various number of channels code all (mostly) follow this pattern:

blank channels
set powers to shapes' 'db' values
set pwrf's to shape's 'a' values
set phases to passed in phase values
delay for 3.0e-6 time
unblank channels
delay for shape1's t1 time
delay for 2.0e-6 time
pulse
delay for shape1's t2 time
unblank channels

I am listing the 1 channel case as the basic example. These last two lines are in reversed order in the 2 channel and 3 channel case and that was the impetus for much of this ticket.

I was always operating under the assumption that blanks and unblanks should balance each other out. But clearly that is not the case in the existing code. Notice how there is one blank but two unblanks.

So is there something about unblanking that resets timings or does something required to the existing state or for the existing flow? Is that a reason we could have 2 unblanks for 1 blank? Or is having 2 unblanks a bug?

bdezonia, 2025-03-19:
@DanIverson would it help if I opened a new ticket on these topics with only the remaining relevant issues outlined? I could then close this mess of a ticket.

bdezonia, 2025-06-19:
@DanIverson can we look at this sometime?