OpenVnmrJ/OpenVnmrJ #876: jexp macro not created in one openvnmrj installation

SanderBash, 2024-06-14:
**Describe the bug**
I have two OpenVNMRJ systems. On one of the systems, when I use 'cexp(expno)' to create an  experiment, it automatically creates a macro (jexp) to move to that experiment. On the other system, only jexp1-9 work, for higher expno's the jexp10, jexp11 etc. do not work. jexp(10) or jexp(11) work, but when using wft, I get an error that the datafile is not open.

**Expected behavior**
I would expect the cexp(expno) command to automatically create jexp macros to move to experiments with expnos larger than 9

**Version of OpenVnmrJ you are using.**
OpenVnmrJ 4.2 from Github

**Environment (please complete the following information):**
 - OS: Almalinux 8

**Additional context**
Both systems were installed with the same Almalinux ISO and OpenVnmrJ files.

DanIverson, 2024-06-14:
On the system that does not work, could you run the following command from a terminal:
   Vnmrbg -mback -n1 "debug('c3') cexp(10)"
and attach the output.

SanderBash, 2024-06-20:
I get the message that hte Vnmrbg command doesn't exist :(

DanIverson, 2024-06-20:
Try running
/vnmr/bin/Vnmrbg -mback -n1 "debug('c3') cexp(10)"
and attach the output

SanderBash, 2024-06-20:
Thanks, that works. I get the message that the environment variable 'vnmruser' is missing.

DanIverson, 2024-06-20:
Could you run makeuser to update the user environment. Log out and then log in again and try the Vnmrbg command again.

SanderBash, 2024-06-21:
I did and attached the outcomes in the .txt files.
[makeuser result.txt](https://github.com/user-attachments/files/15924629/makeuser.result.txt)
[vnmrbg result after makeuser.txt](https://github.com/user-attachments/files/15924630/vnmrbg.result.after.makeuser.txt)

DanIverson, 2024-06-21:
The cexp(10) command failed because exp10 already exists. Can you try one more time. This time, first run
  rm -rf ~/vnmrsys/exp10
and then run the Vnmrbg command.

SanderBash, 2024-06-24:
I again get the environment variable error and I'm not able to resolve it using makeuser this time. I believe I was succesful in removing exp10, it produced no error messages.

DanIverson, 2024-06-24:
Very strange. Could you attach the results from running the following two commands:
  ls -la ~
  env

SanderBash, 2024-07-01:
Hi Dan, not sure what happened, but I tried again to remove exp10 and run vnmrbg, which worked. For now, I'm using cexpnext as a workaround, which works fine somehow.

[vnmrbg after makeuser after rm exp10 succes.txt](https://github.com/user-attachments/files/16054654/vnmrbg.after.makeuser.after.rm.exp10.succes.txt)

DanIverson, 2024-07-02:
The cexpnext macro calls cexp to actually create the experiment. I am going to close this issue. If the problem comes back, we can re-open it.

closed 2024-07-02 by DanIverson