OpenVnmrJ/OpenVnmrJ #886: putvalue function gives error in pulse sequence

SanderBash, 2024-07-28:
**Describe the bug**
Maybe I'm using the function wrong, but I want to set a user parameter ('tc') equal to the acquisition time (at) whenever this user parameter exceeds the value of at.

**To Reproduce**
When compiling the pulse sequence, I get this error:
SHARPER.c: In function 'pulsesequence':
SHARPER.c:40:4: warning: implicit declaration of function 'putvalue'; did you mean 'putwave'? [-Wimplicit-function-declaration]
    putvalue("tc",tc);
    ^~~~~~~~
    putwave
SHARPERdps.o: In function `pulsesequence':
SHARPER.c:(.text+0x49f): undefined reference to `putvalue'
SHARPERdps.o: In function `x_pulsesequence':
SHARPER.c:(.text+0xb47): undefined reference to `putvalue'
SHARPERdps.o: In function `t_pulsesequence':
SHARPER.c:(.text+0x116f): undefined reference to `putvalue'
collect2: error: ld returned 1 exit status

**Expected behavior**
I would expect putvalue to set a parameter to the value of a variable from the pulse sequence.

**Version of OpenVnmrJ you are using.**
OVJ 4.2, running on AlmaLinux 8 in a Virtualbox

bdezonia, 2024-07-28:
Did you mean to call setvalue()?

bdezonia, 2024-07-28:
Or putCmd()?

SanderBash, 2024-07-28:
Actually setvalue("tc",tc); returns the same type of implicit declaration error.

bdezonia, 2024-07-28:
I assume you are missing an include file in your C code. Copy the declarations at the beginning of some stock pulse sequence that ships with OVJ (or can be loaded by compiling one - ahX.c?). Put the same in your pulse sequence file.

SanderBash, 2024-07-28:
That indeed seems like it would be the problem, but most stock sequences I checked just have 'standard.h' included, which mine also does.

bdezonia, 2024-07-28:
Try calling the putCmd(). That is included in abort.h which is one of the 1st files in standard.h. Maybe your code is loading a standard.h from another project from some other place in your path. If putCmd() is not found during compilation/linking then I would either suspect this is the problem or suspect your include call is somehow getting commented out or ifdef'ed out,

SanderBash, 2024-07-28:
I have copied this example code from the manual in my pulse sequence:
d1=1.0;
putCmd("setvalue('d1',%g,'processed')",d1);

That compiles without errors. Can I only use setvalue/putvalue from within putCmd?

bdezonia, 2024-07-28:
We have lots of pulse sequences that call setvalue directly

SanderBash, 2024-07-28:
Do you have any idea how I might fix this problem? Or can you point me to a pulse sequence that uses putvalue or setvalue so I can compare the usage?

bdezonia, 2024-07-28:
I do not know if I can share our code. I suggest you make a 4 linevpulse
sequence. Include standard.h and declare the void pulsesequence routine and
a setvalue() in it. Debug from there. I'm thkingbyou have some include file
from another project getting in the way. Look at your INCLUDE paths.
Rearrange them. Put the vnmrsys or /vnmr include paths  first in your
include path.

bdezonia, 2024-07-28:
OK, I just verified that setvalue is a macro command and not a C command. You can indeed putCmd() with a setvalue() call in it. Sorry for the wild goose chase.

SanderBash, 2024-07-29:
Thanks, I made a minimal pulse sequence:  //test

#include <standard.h>

void pulsesequence()
{

d1=10;

setvalue("d1",d1);

status(A);
delay(d1);
status(B);
rgpulse(pw,0,0,0);
status(C);
}

But still get the implicit declaration error of setvalue. This is my first foray in pulse programming, but I believe I'm following the manual correctly in this regard? I haven't made any changes to the standard header files.

DanIverson, 2024-07-29:
You can only use setvalue from within putCmd. I assume you are using the VnmrJ 4 User Programming manual. It lists putvalue as an available pulse element. The manual is a little less than accurate. The putvalue pulse element, along with putarray and putstring,  was developed by the imaging team. To use it, you must add
  #include "sgl.h"
after the #include <standard.h> statement.  You would also need the imaging appdir turned on. If you are not doing imaging, I would recommend  that you do not include sgl.h and simply wrap the setvalue in the putCmd pulse element. Note that the putvalue pulse element basically uses a series of putCmd calls to do its work.

SanderBash, 2024-07-29:
Thanks, that clears it up. I'll try it with putCmd. And indeed, I was using the inbuilt OVJ 4.2 manual.

DanIverson, 2024-07-29:
Barry,
I thought I might move putvalue, putarray, and putstring from the imaging set of pulse elements to the standard set of pulse elements, i.e., transfer the functions from sglHelp.c to abort.c. They would be declared in abort.h and someone would not need to do the "#include sgl.h" nor turn on the Imaging appdir. I just wanted to confirm that this change would not adversely effect your pulse programming efforts.

SanderBash, 2024-07-30:
Sorry for closing the issue prematurely. When I use putCmd, the value is not set to my user parameter 'tc'. I used create('tc','real','current') to create the parameter. In my pulse sequence I run this code:

   tc = (double) getval("at");
   putCmd("setvalue('tc','%f','current')",tc);
   printf("invalid tc value, tc set to at""\n");

But if at =1 and I set tc = 2 and run dps, tc is still set to tc =2.

DanIverson, 2024-07-30:
The dps command does not execute putCmd statements. You can use go('check') to see if it is working.

SanderBash, 2024-07-30:
Thanks, I read something along those lines in the manual, but did not fully understand it. Running go('check') does not set tc = at unfortunately.

DanIverson, 2024-07-30:
Can you enter the following OVJ commands:
  jFunc(55,'/vnmr/tmp/debug')
  debug('c2')
  go('check')
  debug('C')
and then attach the /vnmr/tmp/debug file

bdezonia, 2024-07-30:
@DanIverson your proposed changes to abort.c for putvalue etc. would not affect us. Feel free to make the changes.

SanderBash, 2024-07-30:
@DanIverson please find the attached debug file.
[debug.txt](https://github.com/user-attachments/files/16433891/debug.txt)

DanIverson, 2024-07-30:
Do you get any warnings or errors when you "seqgen" the pulse sequence? Can you attach the ~/vnmrsys/seqlib/SHARPER.c file (assuming the one you are testing is SHARPER.c)

SanderBash, 2024-07-31:
No, I don't get errors, just a ptsval error when running dps (I'm working on a datastation right now, but have verified yesterday that the sequence runs correct on a spectrometer system). I have attached a version of the sequence for Github and a debug file for this sequence.
[debug.txt](https://github.com/user-attachments/files/16440515/debug.txt)
[SHARPER.txt](https://github.com/user-attachments/files/16440516/SHARPER.txt)

bdezonia, 2024-07-31:
@SanderBash out of curiosity I looked at your pulse sequence code. Note that the code in line 23 is totally undone by the code on line 24. I just want you to be aware in case this is not what you intended.

SanderBash, 2024-07-31:
Barry, you're right, I'm rewriting the sequence to take frequency as a variable instead of time. This line was left over, as in the end I'd like to have both options available, but I still need to write something to accept either input. Thanks for the heads up and checking the code.

DanIverson, 2024-07-31:
Good catch @bdezonia . Sometimes, we make PSG decisions based on whether a parameter is "active" or not. Lines 23 and 24 could be re-written as something like the following:

if (var_active("tc",CURRENT) == 1) { // tc is active
   tc = getval("tc");
} else {  // tc is "not used" or does not exist
   tc = getval("fm") / getval("fc");
}

Note also that if tc == at, the if test at line 29 will be true and it will never evaluate line 33. You may want the test on line 29 to be
  if (tc < at)

Temporarily, could you comment out line 24 and set tc > at  just to see if the putCmd works?

SanderBash, 2024-08-01:
Thanks, that helps. To make tc or fc inactive, do I set myvar = '', or does it work differently?

I have changed the conditional statements and commented out line 24: 

void pulsesequence()
{
   fc = getval("fc");
   fm = getval("fm");
   tc = getval("tc");
   //tc = (1/fc)*((double)fm);
   tcc = tc-2*rof2-2*alfa; //correct delta values by subtracting rof2 and alfa
   initval(1,v3);

   /* calulate loop count for tc values that will result in at least one loop */
   if (tc < at) {
   lc = (at/tc) - (0.25*tc);
   }
   /* if chunk time tc equals aqcuisition time at, set loop count to 0 (only a 90-acquire is performed) */
   else if (tc == at) {
   lc = 0;
   }
   /* if tc is larger than at, tc is invalid and set to at (setvalue function currently not working, bug) */
   else if (tc > at) {
   tc = (double) getval("at");
   putCmd("setvalue('tc','%f','current')",tc);
   printf("invalid tc value, tc set to at""\n");
   }

But still tc is not set to at when tc > at, when I query tc? in the command line

bdezonia, 2024-08-01:
Can you share the pulse sequence C source again? The dump in this email is
confusing.

SanderBash, 2024-08-01:
Sorry, here it is:
[SHARPER_git.txt](https://github.com/user-attachments/files/16461265/SHARPER_git.txt)

DanIverson, 2024-08-01:
To make a numeric parameter "Not acctive", you set it to 'n', eg. tc='n'
To set it active, either use tc='y' or set it to a value, ie., tc=1
Do you see the message "invalid tc value, tc set to at", indicating that the if branch with putCmd is actually called.

DanIverson, 2024-08-01:
Hit the wrong button to close this issue. Can you just put in a putCmd where there is no question it will be called. For example, at line 24, add the line
  putCmd("banner('putCmd called')");

SanderBash, 2024-08-01:
Yes the "invalid tc value... etc" is always being printed. I've added your suggested line at line 24, but no message is printed when I run go('check').

DanIverson, 2024-08-01:
Can you exit OVJ and assuming you have been running go('check') while joined to exp1, then from the command line enter

strace -f -s 128 -o sout Vnmrbg -mback -n1 "go('check')"

and then attach the sout file. Also, just to confirm, did you install OVJ as a Mercury, Inova, or DDR system?

SanderBash, 2024-08-02:
I installed it as a DDR data system (no spectrometer control), but have two other spectrometer control OVJ 4.2 systems connected to a VNMRS. The sout is attached.
[sout.txt](https://github.com/user-attachments/files/16475371/sout.txt)

DanIverson, 2024-08-02:
In this case, PSG aborted because of an invalid value for ptsval. It never got to the putCmd statements. Run config and set the RF Channel Synthesizer to something other than None. Can you then run the strace command again and send the new sout file?

SanderBash, 2024-08-06:
Great! That was indeed the issue, pstval[2] was set to 0, now to 620 and putCmd() works. Thanks a lot for all the support, both of you.

closed 2024-08-06 by DanIverson