OpenVnmrJ/OpenVnmrJ #963: Can this method be removed?

bdezonia, 2024-10-10:
https://github.com/OpenVnmrJ/OpenVnmrJ/blob/699cbbc57259478dd6c73c92cd303ec6bd673df3/src/nvpsg/soliddecshapes.h#L363

When I recursively grep the whole src directory I see this declaration and nobody ever calling it. It does turn up in comments in pieces of the solids pack and biosolidspack manual.

Can we get rid of this stuff?

DanIverson, 2024-10-10:
You should probably check with Chad. It looks like he intended it to be available.

bdezonia, 2024-10-11:
Dan, can you clarify the context for me? How do you know Chad was involved? Without context he might not have any memory of this decision when I bring it up with him.

DanIverson, 2024-10-11:
All of the biosolidspack sequences were either provided by Chad or were derived from a sequence provided by Chad. He is acknowledged in all of the biosolidspack manuals.

bdezonia, 2024-10-11:
OK. I think we should just keep it. Who knows who might be using it. I'm closing this issue.

closed 2024-10-11 by bdezonia