OpenVnmrJ/OpenVnmrJ #979: openvnmrj3.2 bug on seqgen imaging pulse sequences

mosquitozheng, 2024-12-30:
**Describe the bug**
ubuntu: 22.04
installed openvnmrj3.2 version.
Imaging vnmrj interface:
bug: cannot sucessfully compile all imaging pulse sequences. The imaging sequences cannot be sucessfully loaded in the vnmrj interface. For example: if run "seqgen fsems.c", it gaves the below errors:

/vnmr/psg/sglCommon.c: In function 't_calc_mean_nt_tr.part.0':
/vnmr/psg/sglCommon.c:11480:39: warning: '*trvals' may be used uninitialized [-Wmaybe-uninitialized]
/vnmr/psg/sglCommon.c: In function 'x_calc_mean_nt_tr.part.0':
/vnmr/psg/sglCommon.c:11489:39: warning: '*trvals' may be used uninitialized [-Wmaybe-uninitialized]
/vnmr/psg/sglPrepulses.c: In function 'x_create_arterial_spin_label':
/vnmr/psg/sglPrepulses.c:896:38: warning: '*controlpos_694 + _173' may be used uninitialized [-Wmaybe-uninitialized]
  896 |         } else {
      |                                      ^           
/vnmr/psg/sglPrepulses.c:938:38: warning: '*controlpos_718 + _201' may be used uninitialized [-Wmaybe-uninitialized]
  938 |         shapeAslCtrl = shapelist(aslctrl_rf.pulseName,asl_grad.rfDuration,freqAslCtrl,1,0,'s');
      |                            ~~~~~~~~~~^~~~~~~~~~~~
/vnmr/psg/sglPrepulses.c: In function 't_create_arterial_spin_label':
/vnmr/psg/sglPrepulses.c:896:38: warning: '*controlpos_685 + _173' may be used uninitialized [-Wmaybe-uninitialized]
  896 |         } else {
      |                                      ^           
/vnmr/psg/sglPrepulses.c:938:38: warning: '*controlpos_709 + _201' may be used uninitialized [-Wmaybe-uninitialized]
  938 |         shapeAslCtrl = shapelist(aslctrl_rf.pulseName,asl_grad.rfDuration,freqAslCtrl,1,0,'s');
      |                            ~~~~~~~~~~^~~~~~~~~~~~

Same as if "seqgen gems.c":
/vnmr/psg/sglCommon.c: In function 't_calc_mean_nt_tr.part.0':
/vnmr/psg/sglCommon.c:11480:39: warning: '*trvals' may be used uninitialized [-Wmaybe-uninitialized]
/vnmr/psg/sglCommon.c: In function 'x_calc_mean_nt_tr.part.0':
/vnmr/psg/sglCommon.c:11489:39: warning: '*trvals' may be used uninitialized [-Wmaybe-uninitialized]
/vnmr/psg/sglPrepulses.c: In function 'x_create_arterial_spin_label':
/vnmr/psg/sglPrepulses.c:896:38: warning: '*controlpos_694 + _173' may be used uninitialized [-Wmaybe-uninitialized]
  896 |         } else {
      |                                      ^           
/vnmr/psg/sglPrepulses.c:938:38: warning: '*controlpos_718 + _201' may be used uninitialized [-Wmaybe-uninitialized]
  938 |         shapeAslCtrl = shapelist(aslctrl_rf.pulseName,asl_grad.rfDuration,freqAslCtrl,1,0,'s');
      |                            ~~~~~~~~~~^~~~~~~~~~~~
/vnmr/psg/sglPrepulses.c: In function 't_create_arterial_spin_label':
/vnmr/psg/sglPrepulses.c:896:38: warning: '*controlpos_685 + _173' may be used uninitialized [-Wmaybe-uninitialized]
  896 |         } else {
      |                                      ^           
/vnmr/psg/sglPrepulses.c:938:38: warning: '*controlpos_709 + _201' may be used uninitialized [-Wmaybe-uninitialized]
  938 |         shapeAslCtrl = shapelist(aslctrl_rf.pulseName,asl_grad.rfDuration,freqAslCtrl,1,0,'s');
      |                            ~~~~~~~~~~^~~~~~~~~~~~

bdezonia, 2024-12-30:
These look like they are just warnings. Should compile fine (though may not
run right). Where are the errors that are causing the compile to fail?

DanIverson, 2025-03-14:
Fixed by PR #991

closed 2025-03-14 by DanIverson