OpenVnmrJ/OpenVnmrJ #987: Blank OVJ screen after installation

rdsilv1979, 2025-03-05:
Hi, I followed the steps to install the OVJ on Ubuntu and in the end I have only a blank screen.

[image]

DanIverson, 2025-03-05:
From a terminal, can you run
  ovjDiag
and copy the results here.

rdsilv1979, 2025-03-06:
Dear Dan,

This is the result.

Hostname set-l-lm24rpbfa.luna.kuleuven.be should be added to the /etc/hosts file

/vnmr/bin/ovjDiag complete

Rodrigo.

rdsilv1979, 2025-03-06:
This is the /etc/hosts file:

vnmr1@set-l-lm24rpbfa:/etc$ more hosts
127.0.0.1 localhost
127.0.1.1 set-l-lm24rpbfa

# The following lines are desirable for IPv6 capable hosts
::1     ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
172.16.0.2        inova           
172.16.0.4        inovaauto       
172.16.0.43       V-Slim          
172.16.0.51       V-Cryobay       
172.16.0.70       V-Protune       
172.16.0.71       V-Trap          
172.16.0.101      V-AS768         
172.16.0.100      V-AS768Robot

rdsilv1979, 2025-03-06:
The blank screen turned into the following one during the night:

[image]

DanIverson, 2025-03-11:
You need to be able to 
   ping `hostname`
I think the first two lines of /etc/hosts should be
127.0.0.1 localhost set-l-lm24rpbfa.luna.kuleuven.be
# 127.0.1.1 set-l-lm24rpbfa

rdsilv1979, 2025-03-14:
Dear Dan,

The command ovjDiag results doesn't show a result anymore:

[image]

But the status of the ovnmj is still Inactive.

R.

rdsilv1979, 2025-03-14:
[image]

DanIverson, 2025-03-14:
The OVJ interface now looks functional but the acquisition sotware is not running. What happens when you run
  setacq
from a terminal.

rdsilv1979, 2025-03-17:
vnmr1@set-d-lm255512n:~$ setacq

To run /vnmr/bin/setacq you will need to be the system's root user,
or type cntrl-C to exit.

Please enter this system's root user password

Verify NIC setup
Verify NIC  with IP address 172.16.0.1
Network for the spectrometer is configured
 
Available Network Interface Cards (NIC)
Index   Name         IP             Status
1       eno1         10.34.108.81   up    
2       enp3s0       Not configured down  
 
Spectrometer configured on  at 172.16.0.1
 
 
Would you like to re-configure it (y/n) [n]? 
y
 
Available Network Interface Cards (NIC)
Index   Name         IP             Status
1       eno1         10.34.108.81   up    
2       enp3s0       Not configured down  
 
Select the index of the NIC you would like to use.
Alternatively, you may enter a b to sequentially blink
each NIC and select the proper one.
You may also enter a q to quit
Enter index, b, or q to quit (1-2,b,q): 
2
 
Do you want to configure enp3s0 as
1.  IP 172.16.0.1  (preferred)
2.  IP 10.0.0.1
Enter 1, 2, or q to quit (1,2,q): [1] 
1
Selected ethernet device is enp3s0
Selected IP address is 172.16.0.1
Do you wish to continue (y or n)? [y] 
y
# OpenVnmrJ Start
# OpenVnmrJ End
Restarting network
Stopping Acquisition communications.
Stopping bootpd program...
 
Please reboot the console.
Press the Enter key after pressing the console reset button: 

One moment please...

NMR Console software installation complete

Starting Acquisition communications.

************************************************************************
acqcomm  - script to start and stop the acquisition communication system
************************************************************************

acqcomm will start and stop the acquisition communication system.
It replaces the previous "su acqproc" method.
It needs to be run from a terminal window, not the OpenVnmrJ command line.

Options:

    acqcomm
       will start the acquisition communication system if
       it is not running and stop it if it is running

    acqcomm start
       will start the acqisition communication system

    acqcomm stop
       will stop the acqisition communication system

    acqcomm status
       will display whether the acqisition communication system
       is stopped or started

    acqcomm help
       will display this help message

rdsilv1979, 2025-03-17:
I cannot get the system ready. I believe I don't have the serial communication ready.

[image]

DanIverson, 2025-03-17:
It looks like setacq worked. Does
  ping inova
succeed? What is the output of
  ps -ef | grep proc

rdsilv1979, 2025-03-18:
vnmr1@set-d-lm255512n:~$ ps -ef | grep proc
root         734       2  0 Mar17 ?        00:00:00 [irq/16-processor_thermal_device_pci]
root        1963    1954  0 Mar17 ?        00:00:02 ./Expproc
vnmr1       2178    2058  0 Mar17 ?        00:00:00 postgres: checkpointer process   
vnmr1       2179    2058  0 Mar17 ?        00:00:02 postgres: writer process   
vnmr1       2180    2058  0 Mar17 ?        00:00:01 postgres: wal writer process   
vnmr1       2182    2058  0 Mar17 ?        00:00:00 postgres: autovacuum launcher process   
vnmr1       2183    2058  0 Mar17 ?        00:00:01 postgres: stats collector process   
vnmr1       7691    7494  0 Mar17 ?        00:00:00 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -parentBuildID 20250311062550 -prefsHandle 0 -prefsLen 32579 -prefMapHandle 1 -prefMapSize 263901 -sandboxReporter 2 -chrootClient 3 -ipcHandle 4 -initialChannelId {5783ab09-f2db-48f9-8380-8349d2ba251d} -parentPid 7494 -crashReporter 5 -appDir /snap/firefox/5889/usr/lib/firefox/browser 1 socket
vnmr1       7708    7494  0 Mar17 ?        00:00:00 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 32579 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {4fadc316-ca24-41d5-a9c1-a417b0b399f5} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 2 tab
vnmr1       7714    7494  0 Mar17 ?        00:00:08 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -parentBuildID 20250311062550 -prefsHandle 0 -prefsLen 32579 -prefMapHandle 1 -prefMapSize 263901 -sandboxReporter 2 -chrootClient 3 -ipcHandle 4 -initialChannelId {4bc6cac5-d60c-44e3-8769-9fea9065016c} -parentPid 7494 -crashReporter 5 -appDir /snap/firefox/5889/usr/lib/firefox/browser 3 rdd
vnmr1       7748    7494  0 Mar17 ?        00:00:01 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 32720 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {d04ed572-0126-4564-99af-7860803625bf} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 4 tab
vnmr1       8402    7494  0 Mar17 ?        00:00:04 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -parentBuildID 20250311062550 -sandboxingKind 0 -prefsHandle 0 -prefsLen 41027 -prefMapHandle 1 -prefMapSize 263901 -sandboxReporter 2 -chrootClient 3 -ipcHandle 4 -initialChannelId {b354f930-3cee-4184-992a-d04b6a2d62f0} -parentPid 7494 -crashReporter 5 -appDir /snap/firefox/5889/usr/lib/firefox/browser 5 utility
vnmr1       8410    7494  0 Mar17 ?        00:01:40 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37713 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {71c67a96-97bb-446a-af3b-e9c8fa96c36e} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 6 tab
vnmr1       8417    7494  0 Mar17 ?        00:00:13 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37713 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {932ba5a1-c98e-42a9-ad5c-1ee9d85ac32e} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 8 tab
vnmr1       8538    7494  0 Mar17 ?        00:00:26 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37713 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {a2150f38-014f-4dd3-958f-42de4fadca10} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 9 tab
vnmr1       8975    7494  0 Mar17 ?        00:00:04 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37713 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {6df505b0-82ce-4c46-9128-a1da9360182e} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 12 tab
vnmr1       9083    7494  0 Mar17 ?        00:00:04 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37713 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chr [... cut ...]
vnmr1      55830    7494  0 Mar17 ?        00:00:00 /snap/firefox/5889/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 38007 -prefMapHandle 1 -prefMapSize 263901 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250311062550 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {fa66baee-ebe3-4f86-9b0f-d339662ed783} -parentPid 7494 -crashReporter 6 -greomni /snap/firefox/5889/usr/lib/firefox/omni.ja -appomni /snap/firefox/5889/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5889/usr/lib/firefox/browser 183 tab
vnmr1      73041   55151  0 10:23 pts/6    00:00:00 grep --color=auto proc

rdsilv1979, 2025-03-18:
ping inova
PING inova (172.16.0.2) 56(84) bytes of data.
From set-d-lm255512n.luna.kuleuven.be (172.16.0.1) icmp_seq=1 Destination Host Unreachable
From set-d-lm255512n.luna.kuleuven.be (172.16.0.1) icmp_seq=2 Destination Host Unreachable

rdsilv1979, 2025-03-18:
We are using the ethernet port enp3s0. There is a problem to set this device in Ubuntu:
oot@set-d-lm255512n:~# nmcli d
DEVICE  TYPE      STATE                   CONNECTION 
eno1    ethernet  connected               A-zone     
lo      loopback  connected (externally)  lo         
enp3s0  ethernet  unmanaged

rdsilv1979, 2025-03-18:
Hi Dan,

About the internet configuration, I run again the script setacq. This is the outputroot@set-d-lm255512n:/vnmr/bin# ./setacq
Verify NIC setup
Verify NIC enp3s0 with IP address 172.16.0.1
Network for the spectrometer is configured
 
Available Network Interface Cards (NIC)
Index   Name         IP             Status
1       eno1         10.34.108.81   up    
2       enp3s0       172.16.0.1     up    
 
Spectrometer configured on enp3s0 at 172.16.0.1
 
 
Would you like to re-configure it (y/n) [n]? 
y
 
Available Network Interface Cards (NIC)
Index   Name         IP             Status
1       eno1         10.34.108.81   up    
2       enp3s0       172.16.0.1     up    
 
Select the index of the NIC you would like to use.
Alternatively, you may enter a b to sequentially blink
each NIC and select the proper one.
You may also enter a q to quit
Enter index, b, or q to quit (1-2,b,q): 
2
 
Do you want to configure enp3s0 as
1.  IP 172.16.0.1  (preferred)
2.  IP 10.0.0.1
Enter 1, 2, or q to quit (1,2,q): [1] 
1
Selected ethernet device is enp3s0
Selected IP address is 172.16.0.1
Do you wish to continue (y or n)? [y] 
y
/vnmr/bin/setNIC: line 150: ${file}: ambiguous redirect
/vnmr/bin/setNIC: line 160: ${file}: ambiguous redirect
/vnmr/bin/setNIC: line 162: ${file}: ambiguous redirect

** (generate:82412): WARNING **: 11:44:22.069: Permissions for /etc/netplan/01-network-manager-all.yaml are too open. Netplan configuration should NOT be accessible by others.

** (process:82410): WARNING **: 11:44:22.320: Permissions for /etc/netplan/01-network-manager-all.yaml are too open. Netplan configuration should NOT be accessible by others.

** (process:82410): WARNING **: 11:44:22.379: Permissions for /etc/netplan/01-network-manager-all.yaml are too open. Netplan configuration should NOT be accessible by others.
systemd-networkd is not running, output might be incomplete.
Failed to reload network settings: Unit dbus-org.freedesktop.network1.service not found.
Falling back to a hard restart of systemd-networkd.service
Restarting network
Stopping Acquisition communications.
Stopping bootpd program...

DanIverson, 2025-03-18:
What version of Ubuntu are you using (20, 22, or 24)?
What does
   ls -l /etc/netplan
show? Can you attach the /etc/netplan/01-network-manager-all.yaml file.

DanIverson, 2025-03-18:
Could you unzip the attached file and copy the setNIC script contained in the zip file to /vnmr/bin then rerun

  setacq

and see if that fixed the communications.

[acq.zip](https://github.com/user-attachments/files/19323679/acq.zip)

rdsilv1979, 2025-03-21:
Dear Dan,

No, it still do not communicate.

rdsilv1979, 2025-03-24:
No, setacq run with no errors:

nmcli d shows:
DEVICE      TYPE          STATE                               CONNECTION 
eno1          ethernet   connected                       A-zone     
lo                loopback  connected (externally)  lo         
enp3s0      ethernet   connected (externally)  enp3s0

rdsilv1979, 2025-03-25:
ps -ef | grep proc
root         782       2  0 Mar21 ?        00:00:00 [irq/16-processor_thermal_device_pci]
root        3638    3633  0 Mar21 ?        00:00:12 ./Expproc
vnmr1       3694    3674  0 Mar21 ?        00:00:00 postgres: checkpointer process   
vnmr1       3695    3674  0 Mar21 ?        00:00:11 postgres: writer process   
vnmr1       3696    3674  0 Mar21 ?        00:00:08 postgres: wal writer process   
vnmr1       3697    3674  0 Mar21 ?        00:00:04 postgres: autovacuum launcher process   
vnmr1       3698    3674  0 Mar21 ?        00:00:06 postgres: stats collector process   
vnmr1      44164   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -parentBuildID 20250318162831 -prefsHandle 0 -prefsLen 32641 -prefMapHandle 1 -prefMapSize 263989 -sandboxReporter 2 -chrootClient 3 -ipcHandle 4 -initialChannelId {8ce8a582-abbc-42bf-b112-46ca6e4ef669} -parentPid 43964 -crashReporter 5 -appDir /snap/firefox/5917/usr/lib/firefox/browser 1 socket
vnmr1      44179   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 32641 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {a5e51902-838c-411f-8060-8815c29f7510} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 2 tab
vnmr1      44185   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -parentBuildID 20250318162831 -prefsHandle 0 -prefsLen 32641 -prefMapHandle 1 -prefMapSize 263989 -sandboxReporter 2 -chrootClient 3 -ipcHandle 4 -initialChannelId {bb029851-70ea-4835-a733-2150f5574332} -parentPid 43964 -crashReporter 5 -appDir /snap/firefox/5917/usr/lib/firefox/browser 3 rdd
vnmr1      44222   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 32782 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {4142bdf9-66e2-459e-b830-503515df69b2} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 4 tab
vnmr1      44875   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -parentBuildID 20250318162831 -sandboxingKind 0 -prefsHandle 0 -prefsLen 41055 -prefMapHandle 1 -prefMapSize 263989 -sandboxReporter 2 -chrootClient 3 -ipcHandle 4 -initialChannelId {0bef099c-1226-40c8-af7e-c381dc693b1c} -parentPid 43964 -crashReporter 5 -appDir /snap/firefox/5917/usr/lib/firefox/browser 5 utility
vnmr1      44884   43964  0 Mar24 ?        00:00:14 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37835 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {828ff24d-6d59-45bd-b39a-d34ede0bc3ef} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 6 tab
vnmr1      44888   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37835 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {07d2418c-fe3d-4cb1-a27f-c0c56ceb0d4e} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 7 tab
vnmr1      44905   43964  0 Mar24 ?        00:00:13 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 37835 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {14ee8c9d-edf1-46dc-8582-f12a7c406690} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 8 tab
vnmr1      45158   43964  0 Mar24 ?        00:00:01 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 38222 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {65c76e85-5143-4f55-a6f5-95b90a66d7f7} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 10 tab
vnmr1      45275   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 38222 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipc [... cut ...]
vnmr1      45319   43964  0 Mar24 ?        00:00:00 /snap/firefox/5917/usr/lib/firefox/firefox -contentproc -isForBrowser -prefsHandle 0 -prefsLen 38222 -prefMapHandle 1 -prefMapSize 263989 -jsInitHandle 2 -jsInitLen 247456 -parentBuildID 20250318162831 -sandboxReporter 3 -chrootClient 4 -ipcHandle 5 -initialChannelId {fa8e3798-b7a0-4868-bad5-3ac57891146a} -parentPid 43964 -crashReporter 6 -greomni /snap/firefox/5917/usr/lib/firefox/omni.ja -appomni /snap/firefox/5917/usr/lib/firefox/browser/omni.ja -appDir /snap/firefox/5917/usr/lib/firefox/browser 14 tab
root       60212   60201  0 10:59 pts/2    00:00:00 grep --color=auto proc

rdsilv1979, 2025-03-25:
Now, the spectrometer is disconnected.

nmcli c
NAME    UUID                                  TYPE      DEVICE 
A-zone  0e475281-4a09-48e1-8576-e2efd5918543  ethernet  eno1   
lo      34384315-aca2-41ab-a8a8-a7c3ea85bbfa  loopback  lo     
root@set-d-lm255512n:/home/vnmr1# nmcli d
DEVICE  TYPE      STATE                   CONNECTION 
eno1    ethernet  connected               A-zone     
lo      loopback  connected (externally)  lo         
enp3s0  ethernet  disconnected            -- 

nmcli d
eno1    ethernet  connected               A-zone     
lo      loopback  connected (externally)  lo         
enp3s0  ethernet  disconnected            --

rdsilv1979, 2025-03-25:
The Inova Magnet/Sample regulation board is not getting ready status when I reboot it.

For some reason there is no communication with the PC. Maybe, some of the parameters of the ethernet card is not correct. Could it be a firewall problem?

DanIverson, 2025-03-26:
Could you run 
 sudo  tcpdump -i enp3s0 -w bug
Wait about a minute then kill the tcpdump program wirh a cntrl-C.
Send me the bug file.

DanIverson, 2025-03-26:
Note I modified the instructions to reboot the console after starting tcpdump

rdsilv1979, 2025-03-26:
Here the bug file.

[bug.zip](https://github.com/user-attachments/files/19470690/bug.zip)

rdsilv1979, 2025-03-27:
# ps -ef | grep bootpd
root        3706       1  0 Mar26 ?        00:00:00 /usr/sbin/bootpd -s
root      401641    3451  0 08:17 pts/1    00:00:00 grep --color=auto bootpd

# /usr/sbin/bootpd -s
# bootpd: error(3):  bind: Address already in use

# systemctl status bootpd.service
● bootpd.service - Bootp Daemon
     Loaded: loaded (/usr/lib/systemd/system/bootpd.service; enabled; preset: e>
     Active: active (running) since Wed 2025-03-26 16:32:32 CET; 15h ago
   Main PID: 3706 (bootpd)
      Tasks: 1 (limit: 18556)
     Memory: 352.0K (peak: 1.7M)
        CPU: 3ms
     CGroup: /system.slice/bootpd.service
             └─3706 /usr/sbin/bootpd -s

Mar 26 16:32:32 set-d-lm255512n.luna.kuleuven.be systemd[1]: Starting bootpd.se>
Mar 26 16:32:32 set-d-lm255512n.luna.kuleuven.be systemd[1]: Started bootpd.ser>

DanIverson, 2025-03-27:
The bug file provided a hint about what is going wrong but nothing conclusive. Can we try this once again. This time, in one terminal window, enter
  ps -ef | grep bootpd
and note the PID of the process. It is the first number in the output (eg, 3607 in your example above). Then enter
  sudo strace -s 128 -o sbug -p <PID_of_bootpd>
In a second terminal, enter
  sudo tcpdump -i enp3s0 -w tbug
Then reboot the console. Wait 3 minutes and then kill strace and tcpdump with cntrl-C in the respective terminals. Send me the sbug and tbug files.

rdsilv1979, 2025-03-28:
Here the files.

[bug_files.zip](https://github.com/user-attachments/files/19500527/bug_files.zip)

rdsilv1979, 2025-03-28:
1st Terminal:
/# ps -ef | grep bootpd
root        **3706**       1  0 Mar26 ?        00:00:00 /usr/sbin/bootpd -s
root      974791    3451  0 08:28 pts/1    00:00:00 grep --color=auto bootpd
/# sudo strace -s 128 -o sbug -p 3706
strace: Process 3706 attached

2nd Terminal:
/# tcpdump -i enp3s0 -w tbug
tcpdump: listening on enp3s0, link-type EN10MB (Ethernet), snapshot length 262144 bytes

[bug_files.zip](https://github.com/user-attachments/files/19500758/bug_files.zip)

DanIverson, 2025-03-28:
For reasons I do not yet understand, the bootpd program is not being called to assign IP addresses. Do you have a firewall? If so, could you turn it off and try. Could you do
  sudo dmesg > dbug
and send the dbug file

rdsilv1979, 2025-03-31:
We do have firewall, but I asked the IT department to allow the range of the IP address of the Varian.

Please, find attached the file.

[dbug.zip](https://github.com/user-attachments/files/19532531/dbug.zip)

rdsilv1979, 2025-04-02:
I asked the IT department they let the firewall down for tests, I still can't boot Varian with the Ubuntu PC.

In the meantime, our old CentOS 7 PC was repaired and we are able to use it with the Varian. But we would like to transfer all the data to the new Ubuntu PC and retire the old one.

I noticed that if I boot the Varian with the CentOS PC, and simply unplug the ethernet cable from it and connect to the Ubuntu PC, there is connection. The ideal would be to boot the system with the Ubuntu PC only.

DanIverson, 2025-04-03:
My guess is that there is still some process that has been disabled. My suggestion is to re-load Ubuntu, install OVJ, copy the new setNIC script I sent earlier to /vnmr/bin. Run setacq and confirm communications are working. After that, invite your IT group to secure the system, confirming after each step they take that re-booting the console still allows communications.

DanIverson, 2025-04-08:
This re-enforces my belief that there is still some firewall remnant that is preventing communications.

rdsilv1979, 2025-04-09:
I don't know in fact. I asked for them to turn off all the firewall.

Probably, the main issue comes from the management software of the Ubuntu OS installed by the university. I am asking if we could try a new installation of the Ubuntu OS without the management software.

rdsilv1979, 2025-04-14:
The IT techinician is coming today to try to find the issue with the connection. He told that the management software shouldn't interfere with the ethernet connections.

rdsilv1979, 2025-04-14:
Dear Dan,
We found the following while booting the Varian:

[terminal_output_boot_2.odt](https://github.com/user-attachments/files/19736112/terminal_output_boot_2.odt)

R.